sgreg/4chord-midi #15: Create Revision B.1 PCB

sgreg, 2017-08-01:
Revision B PCB needs some minor modifications:

- LCD back light transistor uses wrong component that messes up pin assignment
- move from bipolar transistor to MOSFET along the way
- zero Ohm links won't be necessary anymore
- button foot print is not an ideal fit for buttons in use, either one needs some changing
- UART connector's VCC pin needs a voltage selector for 0V, 3.3V, 5V to keep flexibility with serial connectors (used as debug interface for developers)
- additional rubber foot required on the top side, otherwise board topples when pressing "<" button
- some silk screen adjustments (button text needs to be lower, polarized capacitors lack polarity markings)

sgreg, 2017-08-19:
Everything done, Revision B hardware is now ready \o/

closed 2017-08-19 by sgreg

szwesleyzhao, 2017-08-19:
Cool, do you need turnkey pcb assembly service in china ? We could one stop stop service from prototype to mass production. Here is email; szhwpcba@gmail.com

sgreg, 2017-08-19:
Thanks, but I got everything I need covered for now.